rerun-io/rerun #3698: Use color image as backprojected depth color map

roym899, 2023-10-05:
This was shown in a video in the release notes before (around 0:38 in https://user-images.githubusercontent.com/1220815/223365363-da13585f-3a91-4cb8-a6ef-8a6fadbeb4eb.webm) but doesn't seem to be available.

I couldn't find an issue for this, feel free to close if there is one already.

**Is your feature request related to a problem? Please describe.**
Visualize registered RGB + depth as a point cloud.

Wumpf, 2023-10-06:
I don't think we ever showed the point cloud colored with an rgb image. What is seen in the image is not a backprojection but simply a depth image and a color image on the same camera, where the depth image has depth projection disabled.

But having color mapped depth reprojection is still a logical feature request we haven't written down yet.

Wumpf, 2023-10-06:
Duplicate to #2702

closed 2023-10-06 by Wumpf