rerun-io/rerun #3699: Support Python 3.12

abey79, 2023-10-05:
Python 3.12.0 was recently released (Oct, 2nd) and we should support it.

This is currently blocked by `pyarrow` 3.12 wheel availability:
* https://github.com/apache/arrow/issues/37880

emilk, 2023-10-24:
Still waiting for pyarrow 14 to show up on pypi

abey79, 2023-11-05:
PyArrow 14 is up :tada:

But still soft-blocked on https://github.com/pytorch/pytorch/issues/110436 (we use it in our test suite).

jleibs, 2023-11-06:
Our conda build will need to wait for:
 - https://github.com/conda-forge/arrow-cpp-feedstock/pull/1201

h-vetinari, 2023-11-07:
FYI, in conda-forge, we support python 3.12 on pyarrow 13.x [since](https://github.com/conda-forge/arrow-cpp-feedstock/pull/1181) more than a month already.

PR #4146 (2024-01-05): Add support for Python 3.12
### What

* Bumps arrow to 14.0.2
* Fixes #3699
* Relates to https://github.com/rerun-io/rerun/issues/2445

I added a bunch of fixes to `scripts/run_all.py`, which has a marked tendency to bit rot:
* https://github.com/rerun-io/rerun/issues/4710

Some workaround introduced to deal with PyTorch not 3.12 ready, see:
- https://github.com/rerun-io/rerun/issues/4704

TODO:
- [x] `nox -s tests-3.12` passing
- [x] `nox -s run_all-3.12 --save` passing
- [x] `nox -s roundtrips-3.12` passing
- [x] Solve this:
  ```
  ❯ pixi add pyarrow==14.0.0 
  × could not determine any ava [... cut]
commit 67d6dd8a (2024-01-05): Add support for Python 3.12 (#4146)

### What

* Bumps arrow to 14.0.2
* Fixes #3699
* Relates to https://github.com/rerun-io/rerun/issues/2445

I added a bunch of fixes to `scripts/run_all.py`, which has a marked
tendency to bit rot:
* https://github.com/rerun-io/rerun/issues/4710

Some workaround introduced to deal with PyTorch not 3.12 ready, see:
- https://github.com/rerun-io/rerun/issues/4704

TODO:
- [x] `nox -s tests-3.12` passing
- [x] `nox -s run_all-3.12 --save` passing
- [x] `nox -s roundtrips-3.12` passing
- [x] Solve this:
  ```
  ❯ pixi add pyarrow==14.0.0 
× could not determine  [... cut]
closed 2024-01-05 by jleibs