rerun-io/rerun #3710: TranslationAndMat3x3 object is not iterable

roym899, 2023-10-05:
**Describe the bug**
Logging a TranslationAndMat3x3 fails with the following warning
```
<stdin>:1: RerunWarning: log: TypeError('TranslationAndMat3x3' object is not iterable)
```

**To Reproduce**
```python
import rerun as rr
rr.init("translation", spawn=True)
rr.log("parent/child", rr.TranslationAndMat3x3(translation=[1,0,0]))
```

**Rerun version**
```
rerun_py 0.9.0 [rustc 1.72.1 (d5c2e9c34 2023-09-13), LLVM 16.0.5] x86_64-unknown-linux-gnu 4f56b02, built 2023-10-05T15:37:48Z
```

roym899, 2023-10-05:
Same for `rr.TranslationRotationScale3D`

roym899, 2023-10-05:
Upon reading [this](https://github.com/rerun-io/rerun/pull/3560), I'm guessing this is not actually a bug, i.e., I can't log datatypes directly.

If that's is right, maybe the error message could be improved?

jleibs, 2023-10-05:
Agreed -- we can definitely make that warning better.

closed 2023-10-07 by emilk