rerun-io/rerun #3715: `_send_warning` always raises in strict mode

jleibs, 2023-10-05:
`_send_warning` sounds innocuous enough that we've used it in some places that maybe shouldn't be fatal for users running in strict mode.

We should rename this in a way that makes it clear that this is the behavior and consider adding a less fatal way of notifying users in cases that really should only ever be warnings.

jleibs, 2023-10-05:
Note, this is the cause of a reported error where `AnyValue` raises a TypeError if passed a `None` value.

emilk, 2023-10-09:
I agree with the rename, but I would expect `strict` to always mean "throw instead of warn"

jleibs, 2023-10-09:
> I agree with the rename, but I would expect strict to always mean "throw instead of warn"

Right now this applies only to python warnings.  For example, this code in the batcher produces a warning from rust.
https://github.com/rerun-io/rerun/blob/main/crates/re_log_types/src/data_table_batcher.rs#L228

Should this raise an exception instead?

jleibs, 2023-10-09:
Interesting -- I'm still not fully convinced that all warnings should be panics/exceptions in strict mode, though I agree many of them probably should be.

Still, my preferred interpretation of strict mode is something along the lines of "exceptions for warnings about something that happen as a direct function of user input and for which user could practically change their code to avoid."  Panicking or throwing exceptions for warnings that happen in circumstances that are outside the control of the user, such as data being dropped because a network connection went down, still seems like a bad experience.

Either way, the helper function was renamed in https://github.com/rerun-io/rerun/issues/3716 and I don't have a concrete warning I want to make non-exceptional at the moment, so I'm going to close for now.

closed 2023-10-09 by jleibs

emilk, 2023-10-09:
Very true, connection errors and other things outside the users control should not be affected by strict mode.

(The Rust `strict` mode is only for tests. All errors are already returned using `Result`, always)