rerun-io/rerun #3717: Use `pixi` to replace `just` and many of the Dockerfile layers

jleibs, 2023-10-05:
- https://github.com/rerun-io/rerun/pull/3707 seems very compelling

Pixi should be able to replace all of our current just commands while bringing the added benefit of running things in an environment with managed dependencies. For example, this gives us access to our code-gen dependencies, the `arrow-cpp` we need for building rerun-cpp, etc.

This lets us replace all of the commands in `just` with versions that handle the dep installation first.
This also lets us collapse many of the layers of our Dockerfile into a single `pixi install` invocation.

abey79, 2023-10-24:
Vaguely relevant issue for me: https://github.com/prefix-dev/pixi/issues/189

emilk, 2023-10-24:
We should probably keep our `requirements.txt` files, at least most of them (e.g. for examples)

Wumpf, 2024-04-12:
the only thing we didn't do is to publish rerun on pixi itself, but otherwise we ticked all the boxes now!

closed 2024-04-12 by Wumpf