rerun-io/rerun #3728: Image plane immediately shrinks to too small of a value for pinhole cameras.

jleibs, 2023-10-06:
**Describe the bug**
When creating a scene with a pinhole, the image plane starts off large, which sets the accumulated bounds, and then immediately shrinks to a small value.

**To Reproduce**
```
python docs/code-examples/pinhole_simple.py
```

**Expected behavior**
Image plane should be set to a reasonable value.

**Screenshots**
[image]

**Rerun version**
0.9

emilk, 2023-10-07:
The scene bounding box should ignore the pinhole frustum imho

jleibs, 2023-10-09:
> The scene bounding box should ignore the pinhole frustum imho

Good news: this seems to be the case.
Bad news: I have no idea what is causing the interaction.

closed 2023-10-10 by jleibs