rerun-io/rerun #3731: Accessibility

emilk, 2023-10-07:
We should strive to have some accessibility support in Rerun. The first and simplest step is turning on the AccessKit feature of `eframe`.

I have no idea of how do to proper accessibility for 3D data though.

lunixbochs, 2023-10-13:
> Accessibility Inspector only sees an application's window(s) if the app is run from a proper app bundle, as opposed to just running an executable from Terminal

I'm using rerun-sdk from Python. Accessibility Inspector was still unhappy until I made a skeleton app like this:
```
mkdir -p rerun.app/Contents/MacOS
cp ~/Library/Python/3.9/bin/rerun rerun.app/Contents/MacOS
```

After this, I was able to select buttons in rerun with Accessibility Inspector and see the tree.

I'll note the accessibility performance is much worse than I'd expect. Rechecking with Talon (which also uses egui with accesskit), I see the same laggy response. Not sure what's up there but it should probably go upstream to egui or accesskit. Basically hovering over a UI element with Accessibility Inspector has around a 1 second lag before it's highlighted, while another app like Safari highlights elements immediately.

emilk, 2023-10-13:
Thanks for investigating this. That latency is very bad 😱!

Wumpf, 2025-09-02:
We have accesskit support enabled, but I don't think we plan to do anything on accessibility for 3d things any time soon. I think that's out of scope of this project

closed 2025-09-02 by Wumpf