rerun-io/rerun #3733: No data available if data exists on other timeliens is confusing

Wumpf, 2023-10-07:
Rerun shows simply "no data" for all entities that exist on a timeline that isn't selected right now. In the selection ui there is no indication that there **is** in fact data, just not on the active timeline
Irritatingly, the blueprint view shows _all_ entities just as normal even if there's no data for them available right now.

There's a bunch of improvements that could be made for this:
* blueprint view could already indicate via style and tooltip that an entity is "inactive" due to the "lack of data"
* upon selection it shouldn't just say "no data" but indicate if there's maybe data on another timeline or time, ideally with links
   * challenge here: Each timeline may be set to something else at any point 

Overall, it's right now easy to shoot oneself in the foot by creating a new timeline and then accidentally no longer logging to it. (but even when done intentionally, the additional context may very helpful!)

abey79, 2023-10-09:
Currently, data is identified by a rather nondescript bullet `·` as icon. Maybe we could have a more informative icon there, reflecting availability of data (of course in a way that's consistent between the Blueprint and Streams trees).

abey79, 2023-10-10:
In any case, the tooltip could display a conspicuous note indicating why no data might be displayed.

abey79, 2023-10-11:
After discussing this issue with @martenbjork, we figured that the best course of action was to _not_ do anything further than #3779 because:
- the source/occurrence rate of user confusion over this is not yet well identified
- more generally, the use-cases involving multiple timeline with mutually exclusive data are not well mapped out
- the interaction with related problem are not yet identified (e.g. should we _also_ highlight entities/components that _have_ data logged on the current timeline, but only _after_ the current cursor position?)
- it generally feels like too much design for too little of a problem, given #3779 addressed 90% of the original issue

abey79, 2023-10-11:
Quick illustration of the 3rd bullet point for records, made using #3792
[image]

Here the `hello` entity and its components are italic because they are only logged to another timeline, and everything else is normal. The `hour_hand` stuff is not visible in the space view though, because it's been logged only _after_ the current cursor position. Should it be highlighted as well in the Blueprints tree? Certainly not in the Streams tree. Though question better avoided for now.

closed 2023-10-12 by emilk