rerun-io/rerun #3735: Rust API: Expose `RecordingStreamError`

kpreid, 2023-10-09:
**Describe the annoyance**

My preferred coding style is to **never** use type aliases for `std::result::Result` (such as `rerun::RecordingStreamResult`), but the type `RecordingStreamError` is private. Rust libraries typically make their error types public even if they also offer a `Result`.

This lack of public type might also make it impossible to pass the error through some generic error-handling code that required the error type to be named, but I don't have a practical example of that.

Please make `RecordingStreamError` public (possibly as a struct wrapper if you wish to avoid revealing the enum variants).

**Additional context**

* `rerun 0.9.0`

emilk, 2023-10-09:
Good catch!

There really should be some compiler setting to catch this mistake 🤔

kpreid, 2023-10-09:
Oh, and I could write a PR for this if you like — just let me know whether the enum should be wrapped to be hidden or not as I mentioned above.

emilk, 2023-10-09:
[unreachable-pub](https://doc.rust-lang.org/rustc/lints/listing/allowed-by-default.html#unreachable-pub) does something else - it checks that everything marked `pub` is exported from the crate. What I want is that anything marked `pub` only uses types that are also marked `pub`

emilk, 2023-10-09:
Let's just make `RecordingStreamError` public. Adding wrappers around everything is very annoying at this early stage in the project. A PR would be very welcome!

closed 2023-10-10 by emilk