rerun-io/rerun #3738: Pre-push git hook that checks for typos and runs `lint.py`

emilk, 2023-10-09:
Motivation: save us from waiting until CI notices common problems
Requirements: Must run in at most 2-3 seconds

I think spell checking and most of `lint.py` makes sense. Maybe `lint.py` needs a `--fast`flag to skip some of its slower checks (if we know what those are).

abey79, 2023-10-09:
`ruff` is a good candidate too for a pre-commit hook.

emilk, 2023-10-09:
Let's add something like `just fast-lint` that the pre-push hook calls.

abey79, 2023-10-09:
https://pre-commit.com or something else?

emilk, 2023-10-09:
I don't think there's no need for any new dependencies. `scripts/setup_dev.sh` should just make sure there is a call to `just fast-lint` inside `.git/hooks/pre-push.sample`

closed 2023-10-24 by emilk