rerun-io/rerun #3741: Tracking issue: arrow cleanup & migration away from `arrow2{-convert}`

teh-cmc, 2023-10-09:
Multiple end-goals:
- Use same arrow lib as the rest of the ecosystem, which is where all the bug & perf fixes are actually happening
- Use inifinitely less space to store Arrow metadata (schema deduplication)
   - #1809
- Make it possible to send raw Arrow data to Rerun and have it just work (`RERUN:component_name`)
  - #3360
  - Also frees up usage of Arrow extensions for actual native extensions (e.g. #3004)
- Native integration with `half` for `f16`
- Etc etc

---

TODO (split into sub-issues as needed):
- [ ] Remove all direct uses of `arrow2` (codegen, data{cell,row,table}, `ArrowBuffer`, etc)
  - Related: #2978 
- [ ] Migrate serde-based components (i.e. blueprint stuff) to `arrow1`
  - https://docs.rs/arrow-json/47.0.0/arrow_json/reader/struct.Decoder.html#method.serialize might be all we need
- [ ] Get rid of Arrow extensions everywhere, introduce `RERUN:component_name`  (#3360)
- [ ] Runtime schema registry / dedupe datatypes (#1809)
- [ ] Remove `DataCell::component_name`

On the way there we might hit a few bumps because we have a lot of redundant ad-hoc code that integrates with `polars` (which is built on top of `arrow2`).

The solution to this is to make sure we only integrate with `polars` in one single place: the `Data{Cell,Row,Table}` layer (https://github.com/rerun-io/rerun/issues/1692).
Once that's done, we can remove all ad-hoc polars code everywhere and just build a `Data{Row,Cell,Table}` anytime we want a `polars::Series`/`polars::DataFrame` (#1759).

Internally, the conversion from `DataTable` to `polars::DataFrame` will require a zero-copy tri-stage conversion from `arrow1`->`arrow2`->`polars`.

---

- Supersedes https://github.com/rerun-io/rerun/issues/1805
- Supersedes #2354

emilk, 2024-07-08:
Let's add glue for converting data between `arrow` and `re_arrow2` and we can start this migration piece-wise. We would have double the dependencies for a transitionary period.

Potential roadmap:
* [ ] Write tools for converting data between `re_arrow2` and `arrow`
* [ ] Migrate codegenned serialization
* [ ] Migrate codegenned deserialization
* [ ] Migrate everything else

jleibs, 2024-07-10:
- I believe https://github.com/rerun-io/rerun/issues/6807 also requires bringing in a dependency on `arrow`

teh-cmc, 2024-08-31:
Blocked on:
* https://github.com/apache/arrow-rs/pull/6300

teh-cmc, 2024-09-05:
New blocker:
* https://github.com/apache/arrow-rs/issues/6360

emilk, 2025-01-21:
This is almost done now. What remains is:

* Porting `re_types_builder` (the generated code is all arrow-rs, but the builder itself uses `arrow2::Datatype`)
* IPC serialization (we're hitting a bug in arrow-rs). Look for `SERIALIZE_WITH_ARROW_1` in the code for details
* FFI communication in `rerun_c`

timsaucer, 2025-03-20:
I have added https://github.com/apache/arrow-rs/issues/7315 which I think is the root cause of the serialization error.

emilk, 2025-03-27:
What's left now is mainly porting `re_types_builder`, which should be pretty straight-forward

closed 2025-04-14 by emilk