rerun-io/rerun #3750: Need a way to reset the accumualted bounding box without resetting the scene

jleibs, 2023-10-10:
Many situations can cause the accumulated bounding box to be too large, in particular having the wrong entities in the scene. Once we remove these items, the accumulated bounding box should shrink correspondingly.

jleibs, 2023-10-10:
Related to giving users a better way to recover from problems such as: https://github.com/rerun-io/rerun/issues/3485

teh-cmc, 2023-10-10:
Similarly -- reducing the size of the image plane doesn't seem to shrink the bounding box either.

emilk, 2023-10-10:
> Similarly -- reducing the size of the image plane doesn't seem to shrink the bounding box either.

We should fix it so that the pinhole frustum image plane does not affect the bounding box at all (see other issues).

emilk, 2023-10-10:
Double-clicking to reset the camera should also reset the bounding box

closed 2023-11-28 by Wumpf