rerun-io/rerun #3754: ability to easily turn off logging etc.

Wumpf, 2023-10-10:


emilk, 2023-10-20:
We need both a compile-time and runtime flag

```C++
if (rerun::enabled()) {
     …
}
```

emilk, 2023-10-24:
Let's focus on early-outing from our `log` calls first

jleibs, 2023-10-24:
This warrants further discussion.

This proposal seems to conceive of a global toggle wherein one could start rerun disabled, and then enable later. However, the underlying rust RecordingStreams don't support a toggle -- the enabled/disabled resolution is made at RecordingStream construction time and then persists until a new RecordingStream is created.

The behavior is summarized by:
https://github.com/rerun-io/rerun/blob/a1e084b0aa8e653173bad8352e6a04f31f4e2877/crates/re_sdk/src/recording_stream.rs#L377C9-L377C9

As a simpler alternative to the above proposal I'm going to return and respect the enabled-state of the RecordingStream as created and minimize the overhead on the C++ side when working with a non-enabled RecordingStream.

closed 2023-10-27 by jleibs