rerun-io/rerun #3788: For every component there should be a documented loggable value equivalent to the default

jleibs, 2023-10-10:
Many of our components such as Radii and Label currently have the property that there is no way to explicitly log a value that mimics the behavior of the viewer when no value is logged. This can be frustrating for users when they want to explicitly log data but still get behavior that matches the defaults.

kpreid, 2023-10-10:
One trivial case that could be easily changed, for example: if a `Text` label component is the empty string, then it could be discarded rather than rendered as a background with no text.

Perhaps numeric data could have a rule that NaN = absent? (For cases like `Boxes3D`'s `radii`.)

Wumpf, 2025-03-12:
* https://github.com/rerun-io/rerun/issues/9266

is about the empty label being incorrectly shown.

Otherwise closing this as "done":
* For resetting components we have `clear` methods on archetypes nowadays which allows you to explicitly "unlog" a thing, internally bu logging an empty array which the viewer (!) always treats the same as the value not existing, thus using the built-in default/fallback
* Directly mimicking a default value is ill-defined for the general case: while not always exploited, the way fallbacks work in the viewer they are tied to arbitrary circumstances. E.g. the default image plane distance of a camera depends on the scene bounding box.

closed 2025-03-12 by Wumpf