rerun-io/rerun #3793: Creating new recordings in the Python SDK doesn't result in new recordings in the viewer

nikolausWest, 2023-10-11:
**Describe the bug**
When I create two recordings with the same application id using the python SDK, only a single recording is shown in the viewer.

**To Reproduce**
```python
import rerun as rr
rr.init("test", spawn=True)
rr.init("test", spawn=True)
```

**Expected behavior**
The viewer should show two recordings with the application id "test"

**Screenshots**
It only shows a single recording
[image]

**Rerun version**
0.9

nikolausWest, 2023-10-11:
I put this on the 0.9.1 milestone but it's ok to move if it's too hard to get in in time.

teh-cmc, 2023-10-11:
This works:
```rust
fn main() -> Result<(), Box<dyn std::error::Error>> {
    let rec =
        rerun::RecordingStreamBuilder::new("test").connect(rerun::default_server_addr(), None)?;
    let rec =
        rerun::RecordingStreamBuilder::new("test").connect(rerun::default_server_addr(), None)?;
    Ok(())
}
```

This works:
```cpp
int main() {
    {
        auto rec = rr::RecordingStream("test");
        rec.connect("127.0.0.1:9876").throw_on_failure();
    }

    {
        auto rec = rr::RecordingStream("test");
        rec.connect("127.0.0.1:9876").throw_on_failure();
    }
}
```

This doesn't work:
```py
rr.init("test")
rr.connect()
rr.init("test")
rr.connect()
```

This doesn't work:
```py
rr.init("test", spawn=True)
rr.init("test", spawn=True)
```

teh-cmc, 2023-10-11:
This actually behaves as expected, it's just surprising that the recording ID defaults to the process' authkey: this should be opt-in.

Decision:
- Make sure the current behavior is documented properly for 0.9.1 (TODO: open a ticket)
- Make the authkey logic opt-in for 0.11 (this ticket)

emilk, 2023-10-24:
I suggest we do this at the same time as we refactor `init` overall