rerun-io/rerun #3794: C++ universal "maybeowned" adapting container used everywhere instead of std::vector

Wumpf, 2023-10-11:
The ComponentBatch class introduced in #3791 is not actually specific to components at all. We should build it out to a "universal" container that we use in all codegen wherever we use std::vector right now.
It is also what all serialize methods should work with.

Wumpf, 2023-10-12:
When looking into this we should keep in mind `TensorBuffer` which might be a bit harder to solve but should also use the maybe-owned concept!

emilk, 2023-10-23:
Next step of this is also to support strides

Wumpf, 2023-11-10:
Talking to a user, not only strides would be useful but also byte offsets. Example: Struct like this:

`struct PointAndColor {
float x,y,z;
uint32_t color;
}` 

We want it to be convenient to log arrays of this struct both as points and colors!

Wumpf, 2023-11-17:
The original issue was solved. Split all the other things that came up into different issues and  list them in a tracking issue here
* #4260

closed 2023-11-17 by Wumpf