rerun-io/rerun #3847: SIGSEGV on startup on M1

max-cura, 2023-10-12:
**Describe the bug**
Running the example from the Rust Quick Start page segfaults on 0.9.0, 0.9.1, and 0.10.0-alpha.5.
Running rerun-cli segfaults on 0.9.1, and 0.10.0-alpha.5 (rerun-cli@0.9.0 failed to build), and the current main branch (commit 432d7d2).
In both cases, running the binary from inside lldb results in normal function.

**To Reproduce**
`cargo install rerun-cli@{any of the versions given above}` and then simply `rerun`.
Note that `/usr/bin/lldb rerun` runs without faulting.

The first example from https://www.rerun.io/docs/getting-started/rust can also be used, which will also generate a segfault.

**Expected behavior**
No segfaults on startup.

**Backtrace**
Stack trace `cargo install rerun-cli@0.9.1`, `RUST_LOG=debug rerun`
<details>

```
[2023-10-12T19:47:14Z DEBUG re_memory::memory_limit] Setting memory limit to 24.0 GiB, which is 75% of total available memory (32.0 GiB).
[2023-10-12T19:47:14Z INFO  re_sdk_comms::server] Hosting a SDK server over TCP at 0.0.0.0:9876. Connect with the Rerun logging SDK.
[2023-10-12T19:47:14Z DEBUG eframe] Using the wgpu renderer
[2023-10-12T19:47:14Z DEBUG eframe::native::run] Entering the winit event loop (run_return)…
[2023-10-12T19:47:14Z DEBUG eframe::native::file_storage] Loading app state from "/Users/bosporos/Library/Application Support/rerun/app.ron"…
[2023-10-12T19:47:14Z DEBUG eframe::epi] Failed to decode RON: 1:474: Unexpected missing field `generation` in `SerializedElement`
[2023-10-12T19:47:14Z DEBUG egui_winit::clipboard] Initializing arboard clipboard…
[2023-10-12T19:47:14Z DEBUG re_viewer::native] wgpu adapter name: "Apple M1 Max", device_type: IntegratedGpu, backend: Metal, driver: "", driver_info: ""

Rerun caught a signal: SIGSEGV
Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting
Report bugs: https://github.com/rerun-io/rerun/issues

   2: re_crash_handler::install_signal_handler::signal_handler
   3: _OSAtomicTestAndClearBarrier
   4: <unknown>
   5: <unknown>
   6: <unknown>
   7: <unknown>
   8: <unknown>
   9: <unknown>
  10: <unknown>
  11: <unknown>
  12: <unknown>
  13: <unknown>
  14: <unknown>
  15: eframe::native::epi_integration::EpiIntegration::update
  16: <eframe::native::run::wgpu_integration::WgpuWinitApp as eframe::native::run::WinitApp>::run_ui_and_paint
  17: eframe::native::run::run_and_return::{{closure}}
  18: <winit::platform_impl::platform::app_state::EventLoopHandler<T> as winit::platform_impl::platform::app_state::EventHandler>::handle_nonuser_event
  19: winit::platform_impl::platform::app_state::Handler::handle_nonuser_event
  20: winit::platform_impl::platform::app_state::AppState::cleared
  21: std::panicking::try
  22: winit::platform_impl::platform::observer::control_flow_end_handler
  23: <unknown>
  24: <unknown>
  25: <unknown>
  26: <unknown>
  27: <unknown>
  28: <unknown>
  29: <unknown>
  30: <unknown>
  31: <unknown>
  32: <unknown>
  33: winit::platform_impl::platform::event_loop::EventLoop<T>::run_return
  34: eframe::native::run::with_event_loop
  35: eframe::native::run::wgpu_integration::run_wgpu
  36: eframe::run_native

Rerun caught a signal: SIGSEGV
Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting
[... 159 lines cut ...]
             at eframe-0.23.0/src/native/run.rs:130:9
      std::thread::local::LocalKey<T>::try_with
             at std/src/thread/local.rs:270:16
      std::thread::local::LocalKey<T>::with
             at std/src/thread/local.rs:246:9
      eframe::native::run::with_event_loop
             at eframe-0.23.0/src/native/run.rs:124:16
  36: eframe::native::run::wgpu_integration::run_wgpu
             at eframe-0.23.0/src/native/run.rs:1563:13
      eframe::run_native
             at eframe-0.23.0/src/lib.rs:233:13

Rerun caught a signal: SIGSEGV
Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting
Report bugs: https://github.com/rerun-io/rerun/issues

[1]    90506 segmentation fault  RUST_LOG=debug target/debug/rerun
```
</details>

**Desktop (please complete the following information):**
 - OS: macOS Monterey 12.5 (running on M1 Max)

**Rerun version**
0.9.0, 0.9.1, 0.10.0-alpha.5, main@commit 432d7d2.

nikolausWest, 2023-10-12:
Thanks for the bug report @max-cura, and sorry you had that happen!

Would you mind seeing if you can run through the python quick start on your system?

max-cura, 2023-10-12:
Also segfaults;

I ran `pip3 install rerun-sdk` then ran `python3 -m rerun_demo`, which gave
```
Rerun caught a signal: SIGSEGV
Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting
Report bugs: https://github.com/rerun-io/rerun/issues

Rerun caught a signal: SIGSEGV
Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting
Report bugs: https://github.com/rerun-io/rerun/issues

/opt/homebrew/lib/python3.11/site-packages/rerun_sdk/rerun_demo/__main__.py:24: DeprecationWarning: Please migrate to `rr.log(…, rr.Points2D(…))` or `rr.log(…, rr.Points3D(…))`.
  See: https://www.rerun.io/docs/reference/migration-0-9 for more details.
  rr.log_points("cube", positions=cube.positions, colors=cube.colors, radii=0.5)
```

Wumpf, 2023-10-13:
We tested 0.9.1 on  MacOs 14.0 and 13.4 but I don't think anyone else tried 12.5.
Given that it crashes inside a setting the app icon (`eframe::native::app_icon::AppTitleIconSetter::update`) it's not unlikely that this is about your older Mac version.
I know it's a big ask, but would you mind trying if the problem persists after updating your system?

nikolausWest, 2023-10-13:
Hi again @max-cura, thanks for checking with Python! Before potentially upgrading your OS, perhaps you could try out a simpler eframe based app that also uses a custom app icon like https://crates.io/crates/puffin_viewer just to confirm our suspicion?

max-cura, 2023-10-13:
Yeah, installed puffin_viewer and it indeed has the same pattern (segfault on start, no segfault if started in lldb).
I'll try updating my system tonight.

max-cura, 2023-10-14:
Upgraded to macOS 14.0.

Tried to run the Rust `rerun-cli` again, got this:
```
[2023-10-14T05:12:49Z INFO  re_sdk_comms::server] Hosting a SDK server over TCP at 0.0.0.0:9876. Connect with the Rerun logging SDK.

Rerun caught a signal: SIGBUS
Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting
Report bugs: https://github.com/rerun-io/rerun/issues

   2: re_crash_handler::install_signal_handler::signal_handler
   3: __platform_memmove
   4: <unknown>
   5: <unknown>
   6: <unknown>
   7: <unknown>
   8: <unknown>
   9: <unknown>
  10: <unknown>
  11: <unknown>
  12: <unknown>
  13: <unknown>
  14: <unknown>
  15: eframe::native::epi_integration::EpiIntegration::update
  16: <eframe::native::run::wgpu_integration::WgpuWinitApp as eframe::native::run::WinitApp>::run_ui_and_paint
  17: eframe::native::run::run_and_return::{{closure}}
  18: <winit::platform_impl::platform::app_state::EventLoopHandler<T> as winit::platform_impl::platform::app_state::EventHandler>::handle_nonuser_event
  19: winit::platform_impl::platform::app_state::Handler::handle_nonuser_event
  20: winit::platform_impl::platform::app_state::AppState::cleared
  21: std::panicking::try
  22: winit::platform_impl::platform::observer::control_flow_end_handler
  23: <unknown>
  24: <unknown>
  25: <unknown>
  26: <unknown>
  27: <unknown>
  28: <unknown>
  29: <unknown>
  30: <unknown>
  31: <unknown>
  32: <unknown>
  33: winit::platform_impl::platform::event_loop::EventLoop<T>::run_return
  34: eframe::native::run::with_event_loop
  35: eframe::native::run::wgpu_integration::run_wgpu
  36: eframe::run_native

Rerun caught a signal: SIGBUS
Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting
Report bugs: https://github.com/rerun-io/rerun/issues

[1]    3848 bus error  rerun
```

Error persisted after reinstalling rerun-cli.
Same as before, no error when starting from lldb.

nikolausWest, 2023-10-15:
Thanks for upgrading your system and trying again. This is pretty bad indeed

emilk, 2023-10-19:
I wonder what makes @max-cura:s computer different from mine. Have you tried attaching a debugger @max-cura?

nikolausWest, 2023-10-19:
@emilk: If you read further up on the issue, max wrote that attaching the debugger removed the problem.

Wumpf, 2024-10-22:
@max-cura you're still hitting this I assume? It's been a few version and as far as we know no other user ran into this this 😕

Wumpf, 2025-03-10:
Closing as stale with nothing actionable

However, this might be related to
* https://github.com/rerun-io/rerun/issues/6487

closed 2025-03-10 by Wumpf