rerun-io/rerun #3848: Doc example screenshots don't match output from code

nikolausWest, 2023-10-12:
Here is an example for pinhole + depth image: https://www.rerun.io/docs/reference/types/archetypes/depth_image#depth-to-3d-example

Code
```python
"""Create and log a depth image and pinhole camera."""

import numpy as np
import rerun as rr

depth_image = 65535 * np.ones((8, 12), dtype=np.uint16)
depth_image[0:4, 0:6] = 20000
depth_image[4:8, 6:12] = 45000

rr.init("rerun_example_depth_image", spawn=True)

# If we log a pinhole camera model, the depth gets automatically back-projected to 3D
rr.log(
    "world/camera",
    rr.Pinhole(
        width=depth_image.shape[1],
        height=depth_image.shape[0],
        focal_length=20,
    ),
)

# Log the tensor.
rr.log("world/camera/depth", rr.DepthImage(depth_image, meter=10_000.0))

```

Image in docs
[image]

Reality
[image]

(after manually setting "Backproject Radius Scale")
[image]

In this case, the image looked alot nicer!

Wumpf, 2023-10-12:
looks like someone wanted to simplify the example at some point! Let's give the depth image a higher resolution and take the picture again

teh-cmc, 2023-10-25:
Isn't that done @abey79 ?

abey79, 2023-10-25:
Yes, fixed by #3954

closed 2023-10-25 by abey79