rerun-io/rerun #3896: Pixel center convention not documented

roym899, 2023-10-17:
**Describe the annoyance**
I believe Rerun assumes the pixel center to be at `0.5, 0.5`. OpenCV (mostly?) uses `0.0, 0.0`, Matlab uses `1.0, 1.0` (see e.g., https://se.mathworks.com/help/vision/ref/cameraintrinsicstoopencv.html). It would be nice to document this fact in a few places where it could be relevant. Such as when logging any 2D primitive, images, and pinhole projections. Especially in the latter case the pinhole projection comes with some convention based on how the calibration was performed.

Wumpf, 2023-10-18:
unclear to me though _where_ we document it apart from pinhole where it's relevant. It's a property of the viewer otherwise

rgolovanov, 2024-08-26:
@emilk, @Wumpf  it seems (0.5, 0.5) convention may lead to confusion. For example, when one deals with feature detection algorithms (like corner detector) the visualization of detected features on top of the image isn't aligned well. Point #1 points to exact pixel by rerun visualizes it in the corner and we need to mentally move it to (+0.5, +0.5).

Agree with @Wumpf that it seems to be the Viewer's property and it might be good idea to consider other conventions, e.g. (0, 0) - in this case the 2D view of feature points and theimage will be consistent and automatically aligned.
[image]