rerun-io/rerun #3928: Crash when supplying too large mesh

Wumpf, 2023-10-19:
Similar to too large textures previously, we have issues with too large buffers. Easiest way to get a too large buffer is to supply a too large mesh through any api.

Just as before we need more extensive vetting of user supplied data _and_ need to tell wgpu to allow the largest possible buffer.

emilk, 2023-10-25:
I tested this:

```rs
    rec.log(
        "big",
        &rerun::Mesh3D::new(vec![[0.0, 0.0, 0.0]; 1_000_000_000]),
    )?;
```

and got no crash, at least not on the native viewer.

It seems like the limit is pretty large, and this issue is more theoretical than the texture size limit

Wumpf, 2023-10-26:
@roym899 do you remember how you got this issue? I missed to write the down the exact circumstances

Wumpf, 2023-10-26:
default max buffer size is speced at 256mb https://www.w3.org/TR/webgpu/#limits
so yeah it gotta be a _very_ large mesh, but it can happen :)

Wumpf, 2024-01-25:
Looks like we can also hit this limit when supplying _a lot_ of plot data. The crash then happens when egui tries to resize its staging buffer for uploading the data
```
thread 'ThreadId(1)' panicked at 'Failed to create staging buffer for index data'
egui-wgpu/src/renderer.rs:834
```

PR #5053 (2024-02-06): Fix handling very large meshes & plots (allow maximum available gpu buffers)
### What

We already did this for textures, was overdue to do it for buffers as well.

* fixes #3928
* fixes #5051

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
  * Using newly built examples: [app.rerun.io](https://app.rerun.io/pr/5053/index.htm [... cut]
commit 3a348f88 (2024-02-06): Fix handling very large meshes & plots (allow maximum available gpu buffers) (#5053)

### What

We already did this for textures, was overdue to do it for buffers as
well.

* fixes #3928
* fixes #5051

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
* Using newly built examples:
[app.rerun.io](https://app.rerun.io/pr/5053/in [... cut]
closed 2024-02-06 by Wumpf