rerun-io/rerun #3931: Web-view crashes in chrome on linux

jleibs, 2023-10-19:
**Describe the bug**
When I try to use app.rerun.io on `nightly` I get a crash:
[image]
[image]

**To Reproduce**
Go to https://app.rerun.io/version/nightly/
Try to load any example.

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.10 / Chrome
   - Note: Firefox does not crash

**Rerun version**
- `nightly` only
  - 0.9.1 works without issue

jleibs, 2023-10-19:
https://demo.rerun.io/commit/11dee21/examples/structure_from_motion/ works for me
https://demo.rerun.io/commit/c4e7dc1/examples/structure_from_motion/ appears to crash

Wumpf, 2023-10-19:
As discussed on Slack, the NV12 PR had a different issue which made it fail on all platforms which was fixed in
* #3885
After that we're left with the issue described in the PR description

jleibs, 2023-10-20:
The best working theory we have at the moment is that this is a bad interaction between ANGLE and MESA that occurs in particular edge cases when the shader crosses a size threshold.

The workaround in https://github.com/rerun-io/rerun/pull/3948 lends some support to that theory.

Hopefully this observation can lead us to a more minimal repro and upstream bug-report(s).

emilk, 2023-10-24:
Did https://github.com/rerun-io/rerun/pull/3948 successfully work around the problem?