rerun-io/rerun #3932: Allow adding a 2D image to a 3D space view

emilk, 2023-10-19:
Based on https://discord.com/channels/1062300748202921994/1164158709103083612

Sometimes you want to show a 2D image with some 3D data next to it, like at 1:34 in this video: <https://youtu.be/bKUAZ7IQcf0?si=x6VGZTB8usL90I2v&t=94>

To accomplish this one should be able to do `rr.log("3d/image", rr.Image(…))` + `rr.log("3d/paths", rr.LineStrips3D(paths))`, and then add both entities to the same 3D space view. However, this is currently not allowed:

[image]

jleibs, 2023-10-20:
Interestingly we only recently prevented this, as of: https://github.com/rerun-io/rerun/pull/3822

Rather than treating images as optionally being 3D, I'm inclined to require the explicit addition of an orthographic projection:
```
rr.log("world/image", rr.Image(image))
rr.log("world/points", rr.Points3D([[0, 0, 0], [1, 1, 1], [2, 2, 2]]))
```

jleibs, 2023-10-20:
Github misinterpreted my comment as a resolution