rerun-io/rerun #3937: Fix thread-local shutdown of `RecordingStream` on Mac

emilk, 2023-10-20:
* Original bug: https://github.com/rerun-io/rerun/issues/2889
* Hacky fix: https://github.com/rerun-io/rerun/pull/3929

Calling `RecordingStream::set_thread_local` on Mac used to cause a crash during shudown (https://github.com/rerun-io/rerun/issues/2889). In https://github.com/rerun-io/rerun/pull/3929 we replaced this with an ugly, arbitrary sleep instead.

This sleep has two downsides:
* It adds latency during shutdown to Mac
* It may not be enough to flush the pipe

A better fix would do something like:

```
while !flushed() {
    thread.yield()
}
```

The problem is we cannot use any channels for this, so we would need to rely on atomic bools and similar.

Wumpf, 2023-10-27:
We're seeing the same issue on Windows as well.

emilk, 2023-10-29:
Upstream issue:
* https://github.com/rust-lang/rust/issues/116933