rerun-io/rerun #3941: Exporting visualization for publications and presentations

hovren, 2023-10-20:
As a researcher I not only need to visualize results during development, but I also need to present them to the scientific community.

The type of results I need to show are
- Static plots/images that can be used in printed articles but also on presentations
- Movies for use in presentations or web pages

I would like to see built-in tools for producing publication and presentation grade media.
Some things to consider:
- The simplest solution would be to add controls to capture a playing visualization as a movie, or static images, as it is seen in the viewer.
- A mechanism to control the visualization using code (or configuration) is needed. Maybe you need to visualize different data using the same settings or you simply want to be able to script generation of figures for reproducibility. I envision some kind of API/tools to control e.g. camera views in 3D-plots, zoom-levels for 2D images, etc.
- Where applicable prefer vector-type graphics output (SVG/PDF)

emilk, 2023-10-23:
Related:
* https://github.com/rerun-io/rerun/issues/2305

Nik-V9, 2024-08-13:
Hi, it would be great if this feature can be supported.

I was also wondering if there is a functionality to download images from the viewer?

Wumpf, 2024-08-13:
@Nik-V9 it's possible to download images that were logged previously logged in the upcoming release. But none of the export functionalities described in this ticket are in yet.

Nik-V9, 2024-08-13:
Thanks for the quick response! 

It would be much appreciated if you could consider the above request. Rerun has been amazing for debugging and visualizing 3D stuff. I would love to use it for publication visualizations also!

emilk, 2024-08-13:
I think we could implement this pretty quickly using https://docs.rs/egui/latest/egui/viewport/enum.ViewportCommand.html#variant.Screenshot and then cropping out the content of the space view.

We should probably improve egui to include some sort of description to the screenshot command though (e.g. a `Box<dyn Any>`) that is returned with the screenshot, so we can match the resulting screenshot to the right action ("screenshot _this_ spaceview")

closed 2024-12-03 by emilk