rerun-io/rerun #3982: Rotate image views from

roym899, 2023-10-24:
**Is your feature request related to a problem? Please describe.**
Sometimes incoming images are not rotated "upright". It would be easiest if the image orientation could be modified from within the Rerun viewer similar to how the tensor view can be adjusted.

Rotations in 90 degree steps and mirroring along horizontal and vertical axis would probably cover the most common cases.

Related to https://github.com/rerun-io/rerun/issues/1111

Wumpf, 2023-10-24:
Also related to:
*  #349
*  #683

.. and yeah _almost_ duplicated to
* #1111
It's usecase dependend whether one wants to solve image orientation with a transform or not - depends very much on if anything else in the scene should use the image coordinate system or something else

Wumpf, 2025-06-02:
We should see image/video/etc. orientation as a separate "local" transform and make it directly compatible with EXIF's orientation flag, see http://sylvana.net/jpegcrop/exif_orientation.html