rerun-io/rerun #3984: Unintuitive default split when logging three images

roym899, 2023-10-24:
**Describe the annoyance**
When logging more than three images to separate entity paths, the first image will be in its own pane, with the remaining two sharing an extra grid pane. It's a bit annoying to rearrange into a view where all images have the same size. This seems to only happen when logging exactly three images.

**To Reproduce**
```python
import rerun as rr
import numpy as np

rr.init("test", spawn=True)

rr.log("image1", rr.Image(np.random.rand(100,100,3)))
rr.log("image2", rr.Image(np.random.rand(100,100,3)))
rr.log("image3", rr.Image(np.random.rand(100,100,3)))
# rr.log("image4", rr.Image(np.random.rand(100,100,3)))
```

**Expected behavior**
All images should appear in a consistent way in the viewer (i.e., all taking the same amount of space).

**Additional context**
```
rerun_py 0.10.0-alpha.7+dev [rustc 1.72.1 (d5c2e9c34 2023-09-13), LLVM 16.0.5] x86_64-unknown-linux-gnu main 47414c8, built 2023-10-24T12:00:38Z
```

Wumpf, 2023-10-24:
@emilk something we want to consider fixing in the default tiling or close as won't fix with explicit user blueprint as the future solution?

emilk, 2023-11-06:
I think the problem here is that we have a heuristic for the default arrangement that will never be perfect.

The current heuristic is not smart enough to know that some of the space-views have a fixed aspect ratios (as 2D space-view have, but 3D space-views do not). Adding this support is possible, but challenging.

The real solution is to make it easier for users to change the default from code and in the UI:

* https://github.com/rerun-io/rerun/issues/2276
* https://github.com/rerun-io/rerun/issues/2652