rerun-io/rerun #3985: Images with same parent entity (except root) show up in same image view

roym899, 2023-10-24:
**Describe the annoyance**
By default multiple images appear in one image view alpha-blended when logged to with the same parent entity. It's quite annoying from here to view these images separately.

In the Blueprint + menu such images also only show up bundled by the parent entity path.

Basically I'll manually have to hide the other images if I want to see only one.

**To Reproduce**
```python
import rerun as rr
import numpy as np

rr.init("test", spawn=True)

rr.log("images/image1", rr.Image(np.random.rand(100,100,3)))
rr.log("images/image2", rr.Image(np.random.rand(100,100,3)))
rr.log("images/image3", rr.Image(np.random.rand(100,100,3)))
```

**Additional context**
```
rerun_py 0.10.0-alpha.7+dev [rustc 1.72.1 (d5c2e9c34 2023-09-13), LLVM 16.0.5] x86_64-unknown-linux-gnu main 47414c8, built 2023-10-24T12:00:38Z
```

roym899, 2023-10-24:
I think it might make sense to have different behavior for numbered sequence entity paths such as 
`images/#1`, `images/#2`, `images/#3` -> one pane with tabs
vs 
`images/image_a`, `images/image_b`, `images/image_c` -> separate panes

However, I think the default should never be to alpha blend two images.

Wumpf, 2023-10-24:
we have quite a few examples though where this makes sense. Notably when you have a mask that should be overlayed on another image.

Otherwise this is almost the same issue as, so closing as duplicated
* #2192

roym899, 2023-10-24:
Isn't there a difference between image + (one or multiple segmentation images) and image + depth vs multiple images? I agree that in the first cases the layering makes sense, but in the latter I think this rarely seems like the right choice?

roym899, 2024-02-01:
Reopening after a short discussion with @Wumpf. 

Two cases for which it would be nice to remove this heuristic for multiple normal Images:
- logging multiple images from a dataset to say /dataset/img1, /dataset/img2, ...
- logging stereo images /camera/rgb_left, /camera/rgb_right

Not sure if there are any common cases where the alpha-blending would be preferable?

I think for combinations of Image + DepthImage / SegmentationImage it makes sense to overlay, however, for multiple images with the same parent I'd prefer not to have this as the default.

Wumpf, 2024-02-09:
Might be fixed by
* #5148

need to check

Wumpf, 2024-02-12:
confirmed fixed

closed 2024-02-12 by Wumpf