rerun-io/rerun #3992: Standalone viewer exe opens a command prompt on Windows

Wumpf, 2023-10-24:
* Download Windows viewer executable https://github.com/rerun-io/rerun/releases
* run it not from command line
* get a command line window popping up

[image]

emilk, 2023-10-24:
Solution: put this in main.rs:

> #![cfg_attr(not(debug_assertions), windows_subsystem = "windows")] // hide console window on Windows in release

Wumpf, 2023-10-30:
This suggestion works but then we no longer get any console output at all. So e.g. `rerun.exe --help` wouldn't show anything anymore. This problem is not specific to Rust but to Windows in general. We have to think of a more sophisticated setup to get around this issue. Maybe try this: https://stackoverflow.com/a/19168891

emilk, 2023-11-07:
Does this happen when running `spawn` from the Rust and C++ SDKs? How about the Python SDK?