rerun-io/rerun #4100: Add logging benchmarks for C++/Python/Rust

Wumpf, 2023-10-31:
Keep things super simple and do end-to-end profing: Profile running a benchmark binary that internally has a bunch of test cases. This way we can integrate results from all benchmarks in the same way into our CI generated benchmark stats.

We execute it with different parameters to check for different test cases.
Basic set of test cases we should start with:
* log a lot of scalars with different time
* log a point cloud
* log an image
In all cases (unless configured otherwise) log to a memory recording. (profiling other parts of the flow should be part of a different Rust benchmark)

Since we want to simply time process from spawn to end must make sure that data generation is super fast. Maybe print out additional timings in each language where appropriate - this is harder to integrate into CI graphs, but nice for debugging.

Ideally same data on all SDKs. There might be variations though in logging flow that don't map to each of them.

emilk, 2023-11-01:
Very similar to:
* https://github.com/rerun-io/rerun/issues/3084

Wumpf, 2023-12-12:
We have now benchmarks for Python/Rust/C++ but we still don't upload them

emilk, 2024-01-16:
For reference, here are our performance targets:
* https://github.com/rerun-io/rerun/issues/4423

emilk, 2024-01-16:
We also want to explicitly benchmark logging scalars, including setting a timeline value for each logged scalar, i.e. something like

```py
for frame_nr in range(0, 1_000_000) {
    rr. set_time_sequence("frame", frame_nr)
    rr.log("scalar", rr.TimeSeriesScalar(frame_nr / 1000.0))
}
```

emilk, 2024-01-16:
```
just rs-plot-dashboard --num-plots 10 --num-series-per-plot 5 --num-points-per-series 5000 --freq 1000
```

emilk, 2024-01-16:
We also want to check the memory use in the viewer when we have logged 100M scalars or so, to measure the RAM overhead.

emilk, 2024-01-16:
This is closed when we have an easy way to run benchmarks for all languages, and those results are published (perhaps manually) somewhere publicly.