rerun-io/rerun #4124: Flexible expression language for filtering data & transforming data based on built-in or user-added components

genemerewether, 2023-11-02:
In other GUI data analysis apps (e.g. [https://github.com/finos/perspective/blob/master/README.md#features](Perspective), I can add various (visibility) filters based on component values, or pick a data transformation from component value to color.

**Describe the solution you'd like**

1. Allow the user to select and configure colormaps.
1. Allow using built-in and user-added component values to pick from the colormap.
1. Add an expression tokenizer.
1. Provide built-in and user-added components as symbols in the expression.
1. Create derived data columns, either transiently or adding them as new components.
1. Allow using the derived data to filter what entities are displayed.
1. Allow using the derived data to control color values or other visual attributes.

**Describe alternatives you've considered**
1. Just do the filtering in the client app, before sending over the SDK.
1. Send several possible precomputed color values and/or filter criteria as user-specified components, and select from those in rerun viewer

**Additional context**
[image]
[image]

nikolausWest, 2023-11-03:
Hey @genemerewether! Thanks for opening this issue with the detailed breakdown of what you are looking for, it's very appreciated. I also agree that perspective is a cool tool.

We plan to add a lot of these user defined data transformation and exploration features down the road, but aren't yet set up to do it in the right way. One of the things we need to come much further with first is for instance "blueprint from code", the ability to control layout and visualization settings from code.

genemerewether, 2023-11-06:
@nikolausWest thanks for the reply! I'll stay tuned - excited for these updates.