rerun-io/rerun #4137: Improve the error message when WGPU fails to create an adapter

abey79, 2023-11-03:
The current output is as follows when WGPU fails to create an adapter:

```
$ rerun
[2023-11-03T09:03:29Z INFO  re_sdk_comms::server] Hosting a SDK server over TCP at 0.0.0.0:9876. Connect with the Rerun logging SDK.
[2023-11-03T09:03:30Z INFO  winit::platform_impl::platform::x11::window] Guessed window scale factor: 1.5833333333333333
[2023-11-03T09:03:30Z WARN  wgpu_hal::vulkan::instance] create_instance: ERROR_INCOMPATIBLE_DRIVER
[2023-11-03T09:03:30Z ERROR eframe::native::run] Exiting because of error: Wgpu(NoSuitableAdapterFound) on event Resumed
Error: WGPU error: Failed to create wgpu adapter, no suitable adapter found.

Caused by:
    Failed to create wgpu adapter, no suitable adapter found.
```

At least, a link to https://www.rerun.io/docs/getting-started/troubleshooting#graphics-issues should be provided.

Wumpf, 2023-11-03:
The logging prior to the error is actually useful, but the final error that we panic on should be handled by our side and give more useful information, point to the troubleshooting site, tell you to try GL, etc. etc.
We might need to make improvements to eframe to get to this error.

emilk, 2023-11-03:
If that is the full output, then this is not a panic. Our panic handler already links to the troubleshooter guide, and includes a stacktrace and other stuff.

The problem here is that `eframe::run_native` isn't returning the error, even though it _should_, and that's because winit's `EventLoop::run_return` doesn't, in fact, return, at least not in this situation, for whatever reason. Hopefully the upcoming update to the new winit (which has a new event loop) will fix this issue.

emilk, 2024-08-27:
I believe this is fixed. Please reopen if I'm wrong.

closed 2024-08-27 by emilk