rerun-io/rerun #4142: [CPP SDK] rerun::spawn() is annoying to override default values

kyle-figure, 2023-11-03:
**Describe the annoyance**
Overriding one default parameter in `RecordingStream::spawn` requires passing in all previous parameters:
```c++
// Spawn viewer with flush timeout of 10 seconds:
rec->spawn(9876, "75%", "rerun", std::nullopt, 10.0F).exit_on_failure();
```

**Expected behavior**
Be able to override a single default parameter without providing all previous default values.

**Your goals**
It would be nice to define the configs in a struct with defaults and pass it into the function that way. With this, we would be able to do:
```c++
// Spawn viewer with flush timeout of 10 seconds:
rec->spawn({.flush_timeout_sec=10.0}).exit_on_failure();
```

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04 x86_64

Wumpf, 2023-11-04:
ah good point, a parameter struct might be nicer here indeed! We don't get to use the fancy C++20 (or C99...) struct assignment in our examples for backwards compatibility, but it's still a win!

Wumpf, 2023-11-06:
#4149 ironically doesn't move the `flush_timeout_sec` into the struct because `rerun::spawn` doesn't take it. But everything else and and I added a chrono overload. So you can do now:

```
using namespace std::chrono_literals;
// ...
rec.spawn({}, 4s);
```

PR #4149 (2023-11-06): Replace the many parameters of  `rerun::spawn` / `rerun::RecordingStream::spawn` with a `struct`
### What

* Fixes  #4142

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested [demo.rerun.io](https://demo.rerun.io/pr/4149) (if applicable)
* [x] The PR title and labels are set such as to maximize their usefulness for the next release's CHANGELOG

- [PR Build S [... cut]
commit 1f58d96d (2023-11-06): Replace the many parameters of  `rerun::spawn` / `rerun::RecordingStream::spawn` with a `struct` (#4149)

### What

* Fixes  #4142

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested [demo.rerun.io](https://demo.rerun.io/pr/4149) (if
applicable)
* [x] The PR title and labels are set such as to maximize their
usefulness for the next release's CHANGELOG

- [P [... cut]