rerun-io/rerun #4151: C++ API, DEBUG symbol conflict

adujardin, 2023-11-06:
**Describe the bug**
This is technically not a bug but when trying to integrate rerun C++ API into a pretty big C/C++ codebase (with third-party dependencies) the compilation fails because some file already defined "DEBUG" as a macro. While a questionable choice, it's unfortunately pretty common in C code. The compilation fails at https://github.com/rerun-io/rerun/blob/1f58d96d4ff5b2387da76ebffed8fd32287763ef/rerun_cpp/src/rerun/components/text_log_level.hpp#L54
with an error like:

```
text_log_level.hpp:54:26: error: declaration does not declare anything [-fpermissive]
   54 |             static const TextLogLevel DEBUG;
      |                          ^~~~~~~~~~~~

```

One quick and dirty way to fix or at least to confirm this, is something like this is in the same header:

```
#ifdef DEBUG
#undef DEBUG
#endif
```

Another common solution is to not use "DEBUG" and instead DBG, _DBG, etc

**Desktop:**
 - OS: Ubuntu GCC 9.4, although it's not really relevant here

**Rerun version**
0.10

**Additional context**

Wumpf, 2023-11-06:
Good catch. `DEBUG` as name is an identifier is very unfortunate indeed, we'll have to put something else in there.
DBG and _DBG are defines I've seen in the past as well, so we'll have to do something more else 🤔

emilk, 2023-11-06:
```C
#ifdef DEBUG
#undef DEBUG
#endif
```

would be my preference, though it means include-order is important for our users, as `#include <rerun.h>` would un-define `DEBUG`.

Wumpf, 2023-11-06:
Fixed by https://github.com/rerun-io/rerun/pull/4152

closed 2023-11-06 by Wumpf