rerun-io/rerun #4161: For entities, the list of components should use the `ListItem` style.

abey79, 2023-11-07:


abey79, 2024-05-01:
List Item 2.0 is close to ready for this, but I'm encountering a number of issues:
- `DataUi` implementations makes widespread use of `ui.label(...)` even for `UiVerbosity::Small`, which is bad because it doesn't truncate by default. Instead, we should use `ui.add(egui::Label::new(...).truncate(true))`. Easy but annoying to fix.
- `DataUi` ui very often "captures" hover (e.g. when hovering over the egui label), which "unhovers" the list item itself, for a very distracting effect. `ListItem` deals with this by checking if the content is itself hover, and if so uses a dimmer background for a much better experience. However, `DataUi` doesn't return an `egui::Response`, so the hover information is lots. A possible fix for that would be for list item to check cursor position instead of content response.
- Some `DataUi` implementation of `UiVerbosity::Small` is really busted (e.g. `Rotation3D` in axis/angle mode. Not yet sure how widespread this is.

emilk, 2024-05-02:
> * `DataUi` implementations makes widespread use of `ui.label(...)` even for `UiVerbosity::Small`, which is bad because it doesn't truncate by default. Instead, we should use `ui.add(egui::Label::new(...).truncate(true))`. Easy but annoying to fix.

We should make this a global setting, `ui.style_mut().labels.truncate = Some(true);` or whatever

abey79, 2024-05-02:
Yes, the global truncate setting would be really nice!

abey79, 2024-05-07:
> We could make this a global setting, `ui.style_mut().labels.truncate = Some(true);` or whatever. We could also add a `ui.label_truncated(…)` helper.

https://github.com/emilk/egui/issues/4473

PR #6309 (2024-06-04): Use `list_item` for the component list in `InstancePath::data_ui`
### What

- Fixes #4161
- Follow up to #6297
- Follow up to #6325

This PR uses `list_item` for the component list in the entity path selection panel and tooltip.

~~Blocked by https://github.com/emilk/egui/issues/4471 to resolve the tooltip "first frame flicker"~~

TODO:
-  [x] ⚠️ full_span_scope panic on hover on spatial space view: https://github.com/rerun-io/rerun/issues/6246

#### selection panel

before:

[image]

after:

[image]

*Note*: the hover behaviour is the biggest change here: now full span, previously just on the [... cut]
commit aaf9d5e3 (2024-06-04): Use `list_item` for the component list in `InstancePath::data_ui` (#6309)

### What

- Fixes #4161
- Follow up to #6297
- Follow up to #6325

This PR uses `list_item` for the component list in the entity path
selection panel and tooltip.

~~Blocked by https://github.com/emilk/egui/issues/4471 to resolve the
tooltip "first frame flicker"~~

TODO:
- [x] ⚠️ full_span_scope panic on hover on spatial space view:
https://github.com/rerun-io/rerun/issues/6246

#### selection panel

before:

<img width="519" alt="image"
src="https://github.com/rerun-io/rerun/assets/49431240/afa51449-d7bf-4c0c-9d57-0ac [... cut]
closed 2024-06-04 by abey79