rerun-io/rerun #4185: Ginormous recordings with only a handful of rows obliterate the GC stack

teh-cmc, 2023-11-08:
Our GC generally responds well to stress (see scripts in `tests/python/gc_stress`), but for some reason recordings with one to a few very large rows run into a degenerate case that completely obliterates it.

In practice this ends up looking like this (watch til the end):

https://github.com/rerun-io/rerun/assets/2910679/1f2d6607-932b-44a8-98d7-ecec89c4ee00

emilk, 2023-12-12:
What am I looking at in the video? 😅

teh-cmc, 2023-12-12:
> What am I looking at in the video? 😅

The GC going into a death spiral and drowning under its own weight:
[image]

This might very well be fixed at this point though, I need to try this again with all the improvements that have recently landed.

teh-cmc, 2023-12-14:
I can confirm that:
1. All the stress test suites run much better with the new GC.
2. [many_large_single_row_recordings.py](https://github.com/rerun-io/rerun/blob/main/tests/python/gc_stress/many_large_single_row_recordings.py) still goes into a death spiral, so it's probably something related to adding/dropping recordings themselves.

teh-cmc, 2023-12-18:
I randomly reproduced this while working on primary caching:
[image]

Like, what even is going on here? The store is literally empty, all recordings are gone! Are we leaking some arrow references or something?

teh-cmc, 2024-04-23:
- Might get unblocked by #6066