rerun-io/rerun #4192: Add support to skip the "latest_at" semantics with range queries

abey79, 2023-11-09:
For $REASONS, "empty" range queries exhibit the "latest at" semantics.

In some case, this is questionable:

[image]

In other case, this is outright problematic:

https://github.com/rerun-io/rerun/assets/49431240/206be5af-d117-4879-a550-5d91b5472d5a

Range queres _without_ the "latest at" semantics should possible, and used at least in `TimeSeriesSystem::load_scalars()`.

teh-cmc, 2023-11-10:
Should not be too hard to add a switch for this in `RangeQuery`:
```diff
 pub struct RangeQuery {
     pub timeline: Timeline,
     pub range: TimeRange,
+
+    /// Should the result include the latest compacted state at `range.min`?
+    pub include_latest: bool,
 }
```

teh-cmc, 2024-03-15:
Range queries don't have this weird behavior anymore (they are now multi-tenant proof).

closed 2024-03-15 by teh-cmc