rerun-io/rerun #4194: Support per-space-view-class property defaults

abey79, 2023-11-10:
#4179 highlighted the need for specific default for, in this instance, visible history properties in timeseries space view. This might be a recurring pattern, so a proper mechanism for this should be devised for this (e.g. `SpaceViewClass` should have chance to set default values).

Wumpf, 2024-05-22:
There could be the concept of a default provider that given component name tries to give you a default value or none.
This default provider would be implemented by either `Visualizer` or `SpaceView`. `EditUi` can take in a generic default provider then to determine defaults when needed.

Related to:
* #6071

Wumpf, 2024-06-04:
* Fixed in #6465

closed 2024-06-04 by Wumpf