rerun-io/rerun #4207: Provide CMake install target setup

Wumpf, 2023-11-13:
Ever since we released the C++ SDK, having no cmake INSTALL is a common limitation people hit. Usecase examples:

* Use with a different build system (see https://github.com/kyle-figure/bazel-minimal-rerun/blob/main/third_party/rerun/patches/cmake.patch)
* Not having to build Rerun again from scratch for several rerun using applications on the same machine
* Using it with your own library that uses `install` (https://discord.com/channels/1062300748202921994/1173517222010966047/1173517222010966047)
*  #4130

rishabh-slamcore, 2023-11-13:
Tried cloning and running `scripts/setup_dev.sh`
It fails in `scripts/setup.sh` line `rustup install 1.72.1`
The script assumes `rustup` is available on my vanilla ubuntu 20.04, which is not the case.

Wumpf, 2023-11-13:
@rishabh-slamcore This issue here is specifically about the C++ SDK distribution, not about building Rerun from the repository. For getting started with the repository check https://github.com/rerun-io/rerun/blob/main/BUILD.md#getting-started-with-the-repository - you first have to install rustup. The setup shell scripts aren't in a good shape generally, we gonna soon replace it with a full pixi setup, see 

* #4170 

But worth noting you only need all of this when you want to build Rerun from the repository which is typically only when you want to modify something for a fork (with the goal of either experimenting, making personal enhancements or contributing) 🙂

PR #4326 (2023-11-24): Support cmake install
### What

* Depends on #4313
* Fixes #4207

Support and document use of CMake install.
Minimal example has a switch to use `find_package`. Not rolling it out to others because it's just not as ergonomic, don't want to advertise it all that much. However, the minimal example is used in the sdk package script to test that install is in order before uploading the package script (runs locally). I manually tested the case of a rerun install depend on an arrow system install.

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blo [... cut]
commit f18c4cb9 (2023-11-24): Support cmake install (#4326)

### What

* Depends on #4313
* Fixes #4207

Support and document use of CMake install.
Minimal example has a switch to use `find_package`. Not rolling it out
to others because it's just not as ergonomic, don't want to advertise it
all that much. However, the minimal example is used in the sdk package
script to test that install is in order before uploading the package
script (runs locally). I manually tested the case of a rerun install
depend on an arrow system install.

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/ [... cut]
closed 2023-11-24 by Wumpf