rerun-io/rerun #4211: Freeze + Crash while logging 3d-data

playreplayoliver, 2023-11-13:
**Describe the bug**
Freeze and crash.

**To Reproduce**
It happens quite a lot, but kind of hard to reproduce intentionally.

The setup is:
- Live data streaming to a viewer in python. 
- Feels like it occurs more often when the user interacts with the viewer while continuously logging data in 100-200Hz.

**Backtrace**
```
thread 'ThreadId(1)' panicked at 'Error in Surface::get_current_texture_view: Validation Error

Caused by:
    Parent device is lost

wgpu-0.17.0/src/backend/direct.rs:815
stack backtrace:
   6: core::panicking::panic_fmt
             at core/src/panicking.rs:67:14
   7: wgpu::backend::direct::Context::handle_error_fatal
```
**Desktop:**
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.3 LTS
Release:	22.04
Codename:	jammy

**Rerun version**
rerun_py 0.10.0 [rustc 1.72.1 (d5c2e9c34 2023-09-13), LLVM 16.0.5] x86_64-unknown-linux-gnu release-0.10.0 feea69f, built 2023-10-30T16:58:15Z

**Additional context**
The crash is preceded by a +5s freeze, which reminds me a bit of a previous deadlock-bug in the rerun-viewer.

Wumpf, 2023-11-13:
Device lost usually indicates an issue in the driver. What GPU & driver are you running with?
Did you already get slow-down prior to the issue? The way you describe it (5s freeze) this is likely a TDR (Timeout Detection & Recovery) which happens if something causes too much GPU load. To clarify, is your whole system freezing?

In any case we should figure out how to better handle this - reattempt device creation, dump more information, give better error etc.

Wumpf, 2024-10-22:
@playreplayoliver are you still hitting this? There has been relevant fixes in wgpu by now but tbh I lost track of all the details

playreplayoliver, 2024-10-23:
> @playreplayoliver are you still hitting this? There has been relevant fixes in wgpu by now but tbh I lost track of all the details

No not really. I think we can close this. If I encounter this again i will create another issue.

closed 2024-10-23 by playreplayoliver