rerun-io/rerun #4223: GitHub Action/bot to create Rerun links for attached/linked RRD/files

abey79, 2023-11-14:
What if you could attach a .RRD (zipped, because GH doesn't allow otherwise) to a PR and automatically have a link to a PR-built `app.rerun.io` pointing at said .RRD?

What if users could simply drop a .RRD in their issue and  an `app.rerun.io` a link would automatically show up, using the Rerun version stated in said issue?

.
.
.

What if users anywhere on [GitHub|Discord|Twitter|Mastodon|...] could link/attach any file that's supported by Rerun (natively and/or via an extension) and order a bot to "Open it in Rerun"?

rasmusgo, 2023-11-24:
I was able to load an .rrd file uploaded to github by fiddling with the url a bit: https://app.rerun.io/?url=https://raw.githubusercontent.com/rasmusgo/morten-code/master/data.rrd. The fiddling was to manually follow the redirect and use raw.githubusercontent.com instead of github.com.

abey79, 2023-11-24:
Yeah there are two parts for this:
1) ingest what GitHub accept hosting
2) have a per-PR app.rerun.io build.

As for (1), how did you manage to upload a RRD? GitHub usually refuse anything that's not known extension (including .RRD), thus the idea of supporting .rrd.zip.

emilk, 2024-04-16:
One solution to this is for users to `.zip` the `.rrd`, attach it to a GH issue and have our web viewer read it. See:

* #5541