rerun-io/rerun #4227: Logging of rr.Transform3D very slow

nubertj, 2023-11-14:
Dear rerun team,
Thanks a lot for creating this software and making it available for the public. 
It is really great, and will have a large impact in the future.

**Is your feature request related to a problem? Please describe.**
I am logging different modalities using the Python API of rerun. While everything works well and reliable, me and some co-workers realized that logging the Transform3d via `rr.log(entity_path, rr.Transform3d(rotation, translation, scale))` is much slower than other datatypes such as points, pointclouds or images. In fact, it slows down the entire logging process for us, making it slower than real-time.
This means that for every minute of rerun, we have to wait more considerably more than a minute, effectively slowing down rapid prototyping significantly.

**Describe the solution you'd like**
Having a more performant version of Transform3D logging.

**Describe alternatives you've considered**
I reduced the usage of Transform3D to the bare minimum. Wherever, possible, I am logging points instead of transforms. Unofortunately, in some cases this is not possible, so this Transform3D currently still defines the lower bound of the runtime of the logging script.

jleibs, 2023-11-14:
Thanks for reporting this issue. The Transform3D type has a fair amount of indirection behind it to support all the different permutations of it so I'm not shocked that there's a performance overhead here.

In a very crude test locally I'm seeing about 250us / transform or 4000 Transforms / sec when just logging translation, but I can imagine some additional overhead if I add in quaternions or rotation matrices.

Would you be able to provide any additional information on approximately how many Transform3Ds paths you are logging to and what rate you are logging them at? Just want to make sure we can set up a representative benchmark and drive toward something that might be able to meet your needs.

Also, what style of transform are you trying to log -- I can try to make a proof-of-concept hand-crafted helper for directly building the arrow payload for a specific variant such as Translation + Quaternion, but might as well make it for the specific thing you need.

nubertj, 2023-11-14:
Hi @jleibs,
Thanks a lot for responding so fast. 
Our measurements are at 200Hz, so we would need 200 Transforms/s for being able to perform rerunning in real-time.

If I run a program with 4 threads in Python, at the moment I get around **90 transforms/s** for the loop publishing the Transform3D, while I roughly get **550 points/s** in the same loop publishing Points3D.

These numbers are obtained on an AMD Ryzen™ Threadripper PRO 3975WXs × 64 (although the cores are of course not utilized due to the Python GIL). We are logging the Transform3D with **translation and quaternion**.

nubertj, 2023-11-15:
Quick update, by using **multiprocessing** instead of **threading** I could speed up things  a lot. So now we are 
at **1500 transforms/s for Transform3D** and **7000 points/s for Points3D**.
So the difference is still there, but at least the performance is good now.

jleibs, 2023-11-15:
These numbers still seem off by a pretty large margin.  Here's an updated snippet I used just to log a bunch of translations / quaternions:
```
import time

import numpy as np
import rerun as rr

rr.init("rerun_example_transforms", spawn=True)

arrow = rr.Arrows3D(origins=[0, 0, 0], vectors=[0, 1, 0])

rr.log("base/arrow", arrow)

rand_trans = np.random.rand(1000, 3)
rand_quats = np.random.rand(1000, 4)

start = time.time()
for i in range(1000):
    rr.log(
        "base/arrow",
        rr.Transform3D(
            translation=rand_trans[i],
            rotation=rr.Quaternion(xyzw=rand_quats[i]),
        ),
    )
    
ellapsed = time.time() - start
print(f"Time to log 1000 transforms: {ellapsed:.3f}s. ({1000 / ellapsed:.3f}s transforms / sec)")
```

For me this outputs:
```
Time to log 1000 transforms: 0.297s. (3364.625s transforms / sec)
```

I'm curious what this produces on your machine.

nubertj, 2023-11-15:
This is what I get when running your script:
`Time to log 1000 transforms: 0.541s. (1848.745s transforms / sec)`

jleibs, 2023-11-15:
Your mention of threads got me thinking though so I tested how this works with parallelism:
```
import time

import numpy as np
import rerun as rr
import threading

rr.init("rerun_example_transforms", spawn=True)

arrow = rr.Arrows3D(origins=[0, 0, 0], vectors=[0, 1, 0])

rr.log("base/arrow", arrow)

rand_trans = np.random.rand(1000, 3)
rand_quats = np.random.rand(1000, 4)

def log_transforms(thread_id: int):
    start = time.time()
    for i in range(1000):
        rr.log(
            "base/arrow",
            rr.Transform3D(
                translation=rand_trans[i],
                rotation=rr.Quaternion(xyzw=rand_quats[i]),
            ),
        )

    ellapsed = time.time() - start
    print(f"Thread {thread_id} Time to log 1000 transforms: {ellapsed:.3f}s. ({1000 / ellapsed:.3f}s transforms / sec)")

threads = [threading.Thread(target=log_transforms, args=(i,)) for i in range(4)]
for t in threads:
    t.start()
for t in threads:
    t.join()
```

And now I get:
```
Thread 1 Time to log 1000 transforms: 2.563s. (390.193s transforms / sec)
Thread 2 Time to log 1000 transforms: 2.624s. (381.105s transforms / sec)
Thread 0 Time to log 1000 transforms: 2.637s. (379.278s transforms / sec)
Thread 3 Time to log 1000 transforms: 2.640s. (378.832s transforms / sec)
```

So over a 50% additional reduction in throughput from the threading contention.

jleibs, 2023-11-15:
> This is what I get when running your script: `Time to log 1000 transforms: 0.541s. (1848.745s transforms / sec)`

Awesome, thanks for the context! Will discuss some more with the team tomorrow.

jleibs, 2023-11-15:
Some notes for future investigation:
- When comparing profiling traces between running with 1 thread vs running with 4 threads:
  - I see a ~ 10x performance degradation throughout most the stack.
  - However, for many pyarrow functions, e.g. from_arrays I instead see closer to a 40x performance degradation.

My current working theory is that some lock on this memory pool is causing a bad interaction with the threading model. Likely being further exacerbated by the patterns for how this memory gets handed off to to rust and then later freed via a callback invoked from the batcher thread which would potentially be grabbing the lock without also holding the GIL.

Unfortunately pyarrow doesn't provide facilitites for making this memorypool thread-local, but does provide facilities for allocating them explicitly and then passing them into the arrow functions.
For example StructArray.from_arrays takes an optional MemoryPool argument: https://arrow.apache.org/docs/python/generated/pyarrow.StructArray.html#pyarrow.StructArray.from_arrays

I think the next concrete step (for another day) is to specifically hack in support for passing an optional arrow memory pool into the Archetype constructors, plumbing that down into the _native_to_pa_array interfaces, and updating the transform-related constructors. Or maybe even faster to just hack a thread-local MemoryPool into the transform3d builders directly and bypass any need for plumbing -- if that works then we can come back and figure out the right way of ergonomically giving users more control.

abey79, 2023-11-15:
FWIW, here are some benchmarks I ran based on @jleibs's script, bumped to 100k log calls.

[image]

I used PyCharm's integrated profiler to do the actual profiling, but I believe this is equivalent to using [cProfile](https://docs.python.org/3/library/profile.html). Then I used [flameprof](https://github.com/baverman/flameprof) to generate the flame graph:

```
python -m flameprof file.prof > flame.svg
```

nubertj, 2024-07-16:
Thanks a lot @emilk, I will test it out soon!
FYI @Dvad

closed 2024-08-05 by Wumpf