rerun-io/rerun #4228: Increase Label Size in Box2d

gokcergun, 2023-11-14:
Hi, 

Thanks a lot for the super useful visualizer. I might be stupid but could not find how to increase the size of the labels or a way to add a Text into Boxes2D. Is it possible?

Wumpf, 2023-11-17:
Thanks for filing the issue! Customizing the size of the label isn't possible yet sadly. It for sure though is a kind of customization we'd like to have both on a per space view and even per logged label basis

(Just for the unlikely case that the issue you're facing is overall too small text size: you can zoom the entire application using ctrl/cmd + '+')

gokcergun, 2023-11-18:
Thank you for your answer. Ok,  is it possible to add text to an image?

Wumpf, 2023-11-18:
sadly not something we support yet either. The easiest workaround I can think of is to log a single point with a label and set the point's radius to 0. That way you get a label you can put anywhere you want

Wumpf, 2023-11-18:
Made a few more notes now on this in

* #4266

roym899, 2023-12-07:
Related to #4451, it would be nice if label size could also be set in scene units and UI points similar to points and lines, such that the label takes negligible space when zoomed out, but becomes readable when zoomed in.