rerun-io/rerun #4285: Convert ViewportLayout component to proper LayoutComponent with nested container

jleibs, 2023-11-20:
Right now the entire egui tile tree is serialized as a single field of the "ViewportLayout" component.

However, we want each container to act as its own "entity" in the blueprint store.

Something like:
```
layouts/
  container1/
     .container_type: HORIZONTAL
     .contents -> [container2, spaceview1]
  container2/
     .container_type: VERTICAL
     . conttents -> [spaceview2, spaceview3]

spaceviews/
  ...
```

This means when loading the viewport we need to walk the tree and assemble it into an `egui_tiles::Tree` and similarly at end of frame, if the tree has been modified we need to save it back out as edits to the blueprint.

abey79, 2023-11-23:
Let's keep this related item in mind while working in this area:
- https://github.com/rerun-io/rerun/issues/4322

abey79, 2023-12-15:
Quoting myself in Slack, this actually addresses a fundamental issue:

> Again, for context now that I got clarity of mind about this. The fundamental issue is that we use the egui_tile tree for two purposes: recording our "conceptual" container hierarchy, and to address specific UI needs (in particular draw that little tab thingy at the top of every single space view by forcing a tab container everywhere). These two things are in fundamental contradiction. It's not "just" about fixing egui_tiles, or "just" about having nice blueprint code. It's really a separation of concerns thing.

closed 2023-12-21 by jleibs