rerun-io/rerun #4291: Highlight visible range of time series plot in timeline

Wumpf, 2023-11-21:
Similarly to how visible history highlights a range on the timeline, interacting with a time series plot should do the same for its range.

[image]

abey79, 2024-07-12:
I think this needs a more general treatment:
1) we want to ALWAYS display the visible time range in the time panel WHEN it makes sense
2) since the visible time range can be different on a per-view-entity (aka data result) basis, it actually quite rarely makes sense (because typically lots of views and entities are displayed at the same time)

There is at least one occurence when it makes sense: when we hover over a data result (anywhere in the UI). In that case we have enough context to figure out what the visible time range is, and so we should highlight it in the time panel. This would partially address the original issue: when you hover on a curve, it corresponds to a dataresult, so we should highlight its range.

This feature would be very useful in the context of the data table too.

cc @gavrelina

abey79, 2024-07-12:
Side note: we could also display the View-level visible time range when hovering the view, but I'm not sure that this would be a good idea, since that particular time range might not be used by anything since each entity can override.