rerun-io/rerun #4304: Panel with recent log messages

emilk, 2023-11-22:
All rerun log messages at INFO, WARN and ERROR levels are shown as popup "toasts" to the user. They linger for a few seconds, and then they are gone. It can be pretty annoying if there any many messages in a row, or if a message is big.

We should have a panel where the user can find these messages instead. @martenbjork has a design where we show a little number with unread messages: 
[image]

We should show a scrollback of recent messages, and a red dot next to unread ones.

If we have such a view, then we could change the toast to not show long messages, and never more than one at a time, reducing the amount of noise in the viewer.

### Example
`log::info!("File saved to {path:?}");`

Results in a shortened toast showing: `File saved to C:/Docu…`, and then a white `1` in the title bar, indicating there is one unread INFO-level notification.

teh-cmc, 2024-03-25:
The next logical step once we have a global notifications panel would be to support categorizing warnings/errors per space view, and then add a badge to each space view that only counts the warnings for that specific space view, and redirects to the global notifications panel with a filter preset on click.

jleibs, 2024-05-27:
There is a very related problem to this: In many situations we directly send TextLog messages to the `/rerun` path if a user has logged something that threw a caught-and-handled exception.

Unfortunately, since introducing blueprints, we are no longer guaranteed that the TextLog heuristics will cause a `/rerun` text log view to show up.

It would be nice if this notification mechanism also let you review messages logged to `/rerun` even if you don't have an active TextLog space.

jleibs, 2024-05-27:
This would be a nice feature to have in the Gradio component where a job is running on a remote server and console out isn't immediately visible to the user that kicked off the job in the space.

ShaddyDC, 2024-07-10:
Something like this would be really helpful. I just tried the app for the first time by loading it in the browser and opening a random gltf file I had lying around. It showed me 2 dozen warnings on the right, and before I was able to read even one of them completely, they were all gone again.
Now, in this case, the browser console still had what I believe to be the problem, and it's fine to ignore, but otherwise I'd be a bit concerned if there was anything wrong and I'd be unable to find out.

For context, it was 38 times this:
```
[re_renderer::importer::gltf] re_renderer/src/importer/gltf.rs:250: Textures on meshes are always sampled with a trilinear filter.
 Texture None had Some(NearestMipmapLinear) for min and Some(Linear) for mag filtering, these settings will be ignored
```

jprochazk, 2024-11-18:
Design notes here: https://www.notion.so/rerunio/Error-notification-panel-142b24554b1980c49d9ffce6e8da39ea

First step here is to store recent logs, and implement a popup that displays them, and can be opened by pressing an icon in the top panel:
[image]

emilk, 2024-12-03:
I really want to add the minimal version of this, because the lack of this keeps biting us in the ass.

closed 2024-12-30 by jprochazk