rerun-io/rerun #4307: Tensor view should have hover preview for values just like 2D vie

Wumpf, 2023-11-22:
Essentially this:
[image]

but on the tensor view

Also worth noting that we may want to merge tensor & 2D image view at some point. But this may be longer out.

OskarElek, 2024-10-24:
+1 on @yilun-wu's comment
This feature would be a game changer for me. Any pointers appreciated.

dangom, 2025-02-14:
@yilun-wu not sure, but a quick search suggests the 2D hover is implemented in the [picking ui file of the spatial view.](https://github.com/rerun-io/rerun/blob/0b436dd48d8dca624072695d987e142a8d489f97/crates/viewer/re_view_spatial/src/picking_ui.rs), and then called [here](https://github.com/rerun-io/rerun/blob/0b436dd48d8dca624072695d987e142a8d489f97/crates/viewer/re_view_spatial/src/ui_2d.rs#L228) in the view2d function of the SpatialView2D, which is then used [here](https://github.com/rerun-io/rerun/blob/0b436dd48d8dca624072695d987e142a8d489f97/crates/viewer/re_view_spatial/src/view_2d.rs#L272) in the implementation of the ViewClass for 2D.
Seems an analogous implementation would have to first implement the picking_ui for the tensor, then add it [here](https://github.com/rerun-io/rerun/blob/0b436dd48d8dca624072695d987e142a8d489f97/crates/viewer/re_view_tensor/src/view_class.rs#L239), 
as in what is done for 2D:

```rust
      if let Some(pointer_pos_ui) = response.hover_pos() {
            let picking_context = crate::picking::PickingContext::new(
                pointer_pos_ui,
                scene_from_ui,
                ui.ctx().pixels_per_point(),
                &eye,
            );
            crate::picking_ui::picking(
                ctx,
                &picking_context,
                ui,
                response,
                &mut view_builder,
                state,
                &system_output,
                &ui_rects,
                query,
                SpatialViewKind::TwoD,
            )?;
        } else {
            state.previous_picking_result = None;
        }
```
so that it'd get called [here](https://github.com/rerun-io/rerun/blob/0b436dd48d8dca624072695d987e142a8d489f97/crates/viewer/re_view_tensor/src/view_class.rs#L229), in the implementation of the ViewClass for tensor.

(Take this with a grain of salt, since it's the first time I'm looking at the code, and I now nothing of rust as of yet.)

pablovela5620, 2025-04-14:
Just wanted to include that I'm also in need of this feature! Working on better understanding the attention mechanism and hover would be helpful. As of now I have to print out each batch manually to view the values

[image]

kabouzeid, 2026-01-05:
I'm also running into this a lot and currently have to use a `DepthImage` as a workaround, which is less than ideal.

closed 2026-11-03 by ntjohnson1

ntjohnson1, 2026-11-04:
Landed in 0.42! https://github.com/rerun-io/rerun/releases/tag/0.42.0