rerun-io/rerun #4321: Space View icons should have a consistent style

abey79, 2023-11-23:
Icons should generally encode two piece of information:
- type: space view vs. container vs. entities vs. component vs. group/query
- sub-type: 3D vs. 2d vs. timeseries vs. text etc.

One should immediately recognise a thing as being a space view even if that particular space view type was never seen before.

Currently, this is not the case, eg. here are three space view icons:

[image]

nikolausWest, 2023-12-19:
Note on where these icons will be used:

1. In the blueprint tree
2. In the "contents" section in the selection view when a container is selected
3. In the "add space view or container" modal. Note: these icons may be significantly bigger so it may be preferable to use larger illustrations instead for these.

abey79, 2023-12-20:
### Space View types

Here is the list of existing space views, followed by a few screenshot to illustrate the less obvious ones.

[image]

Bar chart:

[image]

Text document (can be either markdown or raw text):

[image]

Tensor:

[image]

Text Log:

[image]

Dataframe (big table of all the data):

[image]

### Visualiser types

Short-term decision: let's have one icon for all visualiser types.

For context, the visualiser more or less match the list of archetypes, which can be found here: https://www.rerun.io/docs/reference/types/archetypes

### Other icons

Things that are in used today:

[image]

Issues:
- bullet vs. diamond is very hard to tell apart at default scale
- the notion of "group" is likely to disappear
- the notion of "entity-in-a-space-view" (vs. "entity-in-streams") is going to evolve

To sketch it briefly, the blueprint hierarchy will go Container -> Space View -> Queries -> Visualiser -> Data, where Visualiser will display "Data" that results from "Queries". This data will typically be entities from the store, but could also be other things (like inline data). This means that we will shortly need these icons:

Things in the data store:
- entities
- component

Things in the blueprint tree:
- query
- visualizer
- "data" (this is yet ill-defined, there might be subtypes like "data-from-entity", or "static data", etc.)

abey79, 2023-12-20:
Great progresses made today. TBC early Jan 2024. Battle plan here: https://www.figma.com/file/3WX8yUqa8ECmhLwV6NIF3a/Rerun-Design-System?type=design&node-id=1253-9760&mode=design&t=UqJM8rlATmsnLzze-0

abey79, 2024-01-16:
This is mostly done. Remaining effort to be tracked here: https://github.com/rerun-io/rerun/issues/4827

closed 2024-01-16 by abey79