rerun-io/rerun #4325: Support Geo data type

stobias123, 2023-11-23:
would love to be able to plot lat/lon points on a map like openstreetmap. 

**Is your feature request related to a problem? Please describe.**
I'm using the product to track some devices and predictions related to them as they move around a large area. 

**Describe the solution you'd like**
Would love to be able to plot the lat/lon reported by my device over time.

**Describe alternatives you've considered**

**Additional context**

Wumpf, 2023-11-24:
Thanks for filing this! Was surprised we didn't have a ticket for this already!

Related:

* #1460

swarmt, 2023-12-14:
If you only want to be able to display WGS84 coordinates, the simplest way I can think of would be to provide a view origin (lat, lon, alt) that becomes your cartesian origin (0, 0, 0) then use an ENU frame. A good starting point for rust would be [map_3d](https://github.com/gberrante/map_3d).

At smaller scales the ellipsoid to ENU frame separation is probably negligible (~8cm / km).

@stobias123 if you want to display the coordinates now, you could use [pymap3d](https://geospace-code.github.io/pymap3d/) prior to logging to rerun to compute a cartesian coordinate, see `geodetic2enu`.

swarmt, 2023-12-14:
If you want to display map tiles from a WMS, you would need to implement [slippy map](https://en.wikipedia.org/wiki/Tiled_web_map) rendering.

journaux, 2024-05-07:
cesium.js has a native c++ rendering library - might be interesting to integrate

tfoldi, 2024-06-10:
I started to build a custom view for showing OSM/Mapbox maps, I can craft a PR if someone is interested

[image]
[image]

tfoldi, 2024-06-17:
here it is: #6561 

It uses `walkers` map crate (an egui based slippy map) to show map tiles in rerun space view.

journaux, 2024-08-12:
so cool !

emilk, 2024-11-15:
This shipped yesterday in 0.20!

Read more here: https://github.com/rerun-io/rerun/releases/tag/0.20.0

closed 2024-11-15 by emilk