rerun-io/rerun #4337: `About` menu should display the rendering backend being currently used

teh-cmc, 2023-11-27:
It's currently impossible to get that information in many cases AFAICT.

This should ideally tell you the complete story, e.g. "OpenGL ES via wgpu on X11".

teh-cmc, 2024-01-11:
It should also log the actual device being used, especially for linux laptops where it takes a few PhDs to find out what GPU is actually being used by an app.

Wumpf, 2024-01-30:
Loosely related to:

* #4898

emilk, 2024-01-30:
We have this now, though the full details only show on hover:

[image]

And the full details are logged (twice, in fact!):

```
[2024-01-30T11:53:05Z DEBUG egui_wgpu] Picked the only available wgpu adapter: backend: Metal, device_type: IntegratedGpu, name: "Apple M3 Max"
[2024-01-30T11:53:05Z DEBUG re_renderer::context] wgpu adapter backend: Metal, device_type: IntegratedGpu, name: "Apple M3 Max"
```

closed 2024-01-30 by emilk