rerun-io/rerun #4355: Timless counter underflow warning when opening plots example with old blueprint

Wumpf, 2023-11-28:
Got this warning:
```
[2023-11-27T18:13:31Z WARN  re_data_store::time_histogram_per_timeline] Timeless counter underflowed, store events are bugged!
```
when opening the plots demo with an outdated blueprint.

Note that we changed this to be a debug warning on the 0.11.0 release branch

emilk, 2024-01-10:
Let's verify this is a debug-only warning.

Can we repro it?
Can we improve the error message to help figure out what goes wrong?

teh-cmc, 2024-01-10:
> Let's verify this is a debug-only warning.

Ha no, I remember now (by which I mean Andreas pointed that out just above :man_facepalming:): we only changed it on the release branch for 0.11.

Yeah this should really always be a `debug` log, especially now that we enable debug logs by default in debug builds: there's nothing that an end-user can do with that information beside being scared.

Wumpf, 2024-01-16:
so the only action point is to make this a debug log? Okay!

emilk, 2024-01-16:
We haven't seen this in a while 🤷

closed 2024-01-16 by emilk